balena-labs-projects/balena-cam #24: Is this project working on latest balena infrastructure?

svangasse, 2019-01-24:
Hi,
This project looks fantastic, and should be a great starter for a something I'm working on. The problem I'm finding is that it doesn't display video in Chrome from my raspberry pi. The connecting spinner shows for a couple of seconds the disappears. The Chrome logs look positive:

a=candidate:297898e4296c74660bb3ab65d8da90ec 1 udp 1694498815 81.153.71.214 55523 typ srflx raddr 172.18.0.1 rport 55523
a=candidate:0ad8db753210ce9554cf251a9e51e792 1 udp 1694498815 81.153.71.214 32827 typ srflx raddr 172.17.0.1 rport 32827
a=candidate:ea2179218112e69c0463eae46ff5f653 1 udp 1694498815 81.153.71.214 56339 typ srflx raddr 10.114.101.1 rport 56339
a=end-of-candidates
a=ice-ufrag:HGOO
a=ice-pwd:tYQFDadW5na9tgoFBNbIYl
a=fingerprint:sha-256 19:49:12:E9:CF:AE:EA:F1:D0:66:94:F7:31:BC:DC:88:4D:FC:85:A2:76:AA:10:06:E5:34:EE:B1:D9:28:24:2B
a=setup:active
client.js:12 checking
client.js:22 stable
client.js:26 incoming track
client.js:29 Video element added.
client.js:12 connected
client.js:12 completed

and the logs in Balena Cloud also don't show that anthing is wrong:

24.01.19 16:41:05 (+0000)  balena-cam  +++++++ Frame Capture Testing +++++++
24.01.19 16:41:16 (+0000)  balena-cam  Min:  0.999051123000001  seconds
24.01.19 16:41:16 (+0000)  balena-cam  Max:  1.6828183119999984  seconds
24.01.19 16:41:16 (+0000)  balena-cam  Avg:  1.0680984991000013  seconds
24.01.19 16:41:19 (+0000)  balena-cam  +++++++ VideoFrame Generation Testing +++++++
24.01.19 16:41:22 (+0000)  balena-cam  Min:  0.18706517700000092  seconds
24.01.19 16:41:22 (+0000)  balena-cam  Max:  0.1999926510000023  seconds
24.01.19 16:41:22 (+0000)  balena-cam  Avg:  0.1901715952999993  seconds
24.01.19 16:41:27 (+0000)  balena-cam  ======== Running on http://0.0.0.0:3000 ========
24.01.19 16:41:27 (+0000)  balena-cam  (Press CTRL+C to quit)

I followed the instructions and everything seemed to go smoothly. I'm just wondering whether this has been tested recently.

Any help would be very much appreciated.

Thanks,

Steve

mbalamat, 2019-01-25:
Hi @svangasse thanks for trying out balena-cam,

can I have more info about the equipment you used (RPi board, camera etc).
Have you tried refreshing the page? (what was the outcome?)
Also when the connection gets completed if you check the webrtc internals (chrome://webrtc-internals) do you see any traffic on the section about the video?

mbalamat, 2019-01-25:
Hi @svangasse thanks for trying out balena-cam,

I 've tried it about a week ago and it worked as expected (my setup: RPi 3 model B+, official picamera).
Please answer the following questions: 
1. What was the equipment you used (RPi board, camera etc).
2. Have you tried refreshing the page? (what was the outcome?)
3. Have you tried it on firefox or chrome mobile?
4. When the connection gets completed if you check the webrtc internals (chrome://webrtc-internals) do you see any traffic on the section about the video?
5. What is the output of chrome's console if you run 
> video.currentTime

on the page twice (does it change the second time)?

svangasse, 2019-01-28:
Thanks for getting back to me @mbalamat.

1. Same as you: RPi 3 B+ with official camera V2

2. Refreshing the page had no effect - same loading spinner, then nothing:
[image]

3. I've tried it on latest Chrome on my macbook (Version 71.0.3578.98), Safari on macbook, and chrome and firefox on my Android. Same white empty video element as screenshot above

4. There seems to be video data coming over the network:
[image]

5. Running video.currentTime in chrome's console always returns 0:
[image]

I've successfully had still images and video working using the [balena-rpi-python-picamera](https://github.com/balena-io-projects/balena-rpi-python-picamera) project so the hardware seems to be working okay.

svangasse, 2019-01-28:
Interestingly, I just installed Firefox (64.0.2 (64-bit)) on my macbook and it works! So I'm not sure why it's not working on the other browsers

closed 2019-02-05 by mbalamat