balena-labs-projects/balena-cam #38: libjasper-dev

dmiltiades, 2019-07-23:
Problem persists

: Unable to locate package libjasper-dev

mbalamat, 2019-07-24:
Hi @dmiltiades can you verify that you are on commit `82e1b92298396e6c356e79fd230d27e4dc45bf08`.

dmiltiades, 2019-07-24:
Hello Marios,I tried the new commit last night and it did not work unfortunately. The same problem persists.Demetris

dmiltiades, 2019-07-24:
Hello Marios,

maybe I need a guide how to ensure I am cloning the correct commit?

I tried git checkout but I am not succeeding for some reason...i get warnings about ambiguous names..

mbalamat, 2019-07-25:
@dmiltiades you can use `git rev-parse HEAD`

mbalamat, 2019-07-25:
you can use `git pull origin master` to pull the latest changes

mbalamat, 2019-07-25:
@dmiltiades you can always delete the directory and follow the tutorial video here https://www.balena.io/blog/build-a-raspberry-pi-based-network-camera/#pushthecodetothedevice

dmiltiades, 2019-07-25:
warning: refname '82e1b92298396e6c356e79fd230d27e4dc45bf08' is ambiguous.
Counting objects: 531, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (268/268), done.
Writing objects: 100% (531/531), 169.40 KiB | 56.46 MiB/s, done.
Total 531 (delta 256), reused 531 (delta 256)
remote: Resolving deltas: 100% (256/256), done.

[Info]        Starting build for guava, user gh_dmiltiades
[Info]        Dashboard link: https://dashboard.balena-cloud.com/apps/1461584/devices
[Info]        Building on arm03
[Info]        Pulling previous images for caching purposes...
[Success]     Successfully pulled cache images
[balena-cam]  Step 1/10 : FROM balenalib/raspberrypi3-64-debian:stretch
[balena-cam]   ---> 0049024a5b32
[balena-cam]  Step 2/10 : RUN echo "deb http://deb.debian.org/debian jessie main" >> /etc/apt/sources.list
[balena-cam]  Using cache
[balena-cam]   ---> 929fb4502a6c
[balena-cam]  Step 3/10 : RUN apt-get update &&   apt-get install -yq     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     python3     python3-dev     python3-pip     python3-setuptools     libsrtp2-dev     libopencv-dev     libatlas3-base     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*
[balena-cam]   ---> Running in 48d8a999cca9
[balena-cam]  Ign:1 http://deb.debian.org/debian stretch InRelease
[balena-cam]  Get:2 http://security.debian.org/debian-security stretch/updates InRelease [94.3 kB]
[balena-cam]  Get:3 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
[balena-cam]  Ign:4 http://deb.debian.org/debian jessie InRelease
[balena-cam]  Get:5 http://deb.debian.org/debian stretch Release [118 kB]
[balena-cam]  Get:6 http://deb.debian.org/debian jessie Release [77.3 kB]
[balena-cam]  Get:7 http://deb.debian.org/debian stretch Release.gpg [2434 B]
[balena-cam]  Get:8 http://deb.debian.org/debian jessie Release.gpg [1652 B]
[balena-cam]  Get:9 http://security.debian.org/debian-security stretch/updates/main arm64 Packages [480 kB]
[balena-cam]  Get:10 http://deb.debian.org/debian stretch-updates/main arm64 Packages [27.3 kB]
[balena-cam]  Get:11 http://deb.debian.org/debian stretch/main arm64 Packages [6924 kB]
[balena-cam]  Fetched 7816 kB in 3s (2548 kB/s)
[balena-cam]  Reading package lists...
[balena-cam]  Reading package lists...
[balena-cam]  Building dependency tree...
[balena-cam]  Reading state information...
[balena-cam]  E: Unable to locate package libjasper-dev
[balena-cam]  
[balena-cam]  Removing intermediate container 48d8a999cca9
[Info]        Uploading images
[balena-cam]  The command '/bin/sh -c apt-get update &&   apt-get install -yq     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     python3     python3-dev     python3-pip     python3-setuptools     libsrtp2-dev     libopencv-dev     libatlas3-base     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
[Success]     Successfully uploaded images
[Error]       Some services failed to build:
[Error]         Service: balena-cam
[Error]           Error: The command '/bin/sh -c apt-get update &&   apt-get install -yq     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     python3     python3-dev     python3-pip     python3-setuptools     libsrtp2-dev     libopencv-dev     libatlas3-base     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
[Error]       Not deploying release.

remote: error: hook declined to update refs/heads/master
To git.balena-cloud.com:gh_dmiltiades/guava.git
 ! [remote rejected] master -> master (hook declined)
error: failed to push some refs to 'gh_dmiltiades@git.balena-cloud.com:gh_dmiltiades/guava.git'

mbalamat, 2019-08-02:
@dmiltiades that's weird... can you try deleting the directory you cloned and re-clone and re-push?

mbalamat, 2019-08-02:
Also I'd like some more information about your setup. Raspberry Pi model and version, camera module used and what BalenaOS version your device is running (you can find that on your balena-cloud dashboard)

idoodler, 2019-08-04:
@mbalamat I do encounter the same issue. I am using a `Raspberry Pi Zero W` with `balenaOS 2.32.0+rev1`. I freshly cloned the repository and deployed the project using `balena push <PROJECT_NAME>`.

dmiltiades, 2019-08-05:
> Also I'd like some more information about your setup. Raspberry Pi model and version, camera module used and what BalenaOS version your device is running (you can find that on your balena-cloud dashboard)

Raspberry Pi 3 (using 64bit OS) (BETA)
balenaOS 2.38.0+rev1 SUPERVISOR VERSION 9.15.7
Raspberry Pi Camera Module V2 8Mp

I tried deleting the directory too and recloning the same and a different directory which both failed

mbalamat, 2019-11-17:
@dmiltiades I missed this ^^ because it didn't contain any pings, sorry. I'll close this issue. I'd recommend updating your device with the latest balenaOS and retrying.

closed 2019-11-17 by mbalamat