balena-labs-projects/balena-cam #40: push error

jnragu, 2019-08-01:
When pushing, the remote build fails with the error:
`ERROR: Could not find a version that satisfies the requirement aiohttp_basicauth==0.1.0 (from versions: 0.1.3)
ERROR: No matching distribution found for aiohttp_basicauth==0.1.0
The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.0 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1`

jnragu, 2019-08-01:
In the Dockerfile.template I changed the line: 
`RUN pip3 install aiohttp aiohttp_basicauth==0.1.0 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple`
to
`RUN pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple`

It was able to build and upload but now I get "Video device is not ready. Trying to load bcm2835-v4l2 driver"

mbalamat, 2019-08-02:
@jnragu can you verify that you've set up the following environment variables `BALENA_HOST_CONFIG_start_x` to `1` and the gpu_mem vars as described on the README?

mbalamat, 2019-08-02:
@jnragu can you try out the latest master, I've added some extra steps on README to avoid having a default gpu_mem variable on the dashboard messing with the gpu_mem_{256, 512, 1024}

closed 2019-08-05 by mbalamat

jnragu, 2019-08-06:
I do have the environment variables `BALENA_HOST_CONFIG_start_x`to `1`, but I change the gpu_mem according to the README instead of inside of the dashboard fleet variables

russellblake, 2020-07-05:
@mbalamat I am getting the same error and tried all of the suggestions in this post, but I am still getting the error, see attached. Any thoughts on what to try next?
[image]

mbalamat, 2020-07-06:
@russellblake what device are you using? Pi3, Pi4, Pi0 or else?

russellblake, 2020-07-07:
@mbalamat I am using a balenaFin v1.0 with Raspberry Pi Compute Module 3 Plus (CM3+L).