balena-labs-projects/balena-cam #42: Can not push project to balenaCloud

idoodler, 2019-08-04:
I recently pulled this project and tried to deploy it on a `Raspberry Pi Zero W` running `balenaOS 2.32.0+rev1`.

However I am unable to push this project using `balena push <PROJECT_NAME>`

<details><summary>Log from balena push</summary>
<p>

```shell
[Info]        Starting build for ImgFetcher, user gh_idoodler
[Info]        Dashboard link: https://dashboard.balena-cloud.com/apps/1498507/devices
[Info]        Building on arm01
[Info]        Pulling previous images for caching purposes...
[Success]     Successfully pulled cache images
[balena-cam]  Step 1/10 : FROM balenalib/raspberry-pi-debian:stretch
[balena-cam]   ---> b5c859a71dc3
[balena-cam]  Step 2/10 : RUN echo "deb http://deb.debian.org/debian jessie main" >> /etc/apt/sources.list
[balena-cam]  Using cache
[balena-cam]   ---> db9231a20254
[balena-cam]  Step 3/10 : RUN apt-get update &&   apt-get install -yq     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     python3     python3-dev     python3-pip     python3-setuptools     libsrtp2-dev     libopencv-dev     libatlas3-base     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*
[balena-cam]   ---> Running in f2757c715856
[balena-cam]  Ign:1 http://deb.debian.org/debian jessie InRelease
[balena-cam]  Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
[balena-cam]  Get:3 http://deb.debian.org/debian jessie Release [77.3 kB]
[balena-cam]  Get:4 http://archive.raspbian.org/raspbian stretch InRelease [15.0 kB]
[balena-cam]  Get:5 http://deb.debian.org/debian jessie Release.gpg [1652 B]
[balena-cam]  Get:6 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [45.0 kB]
[balena-cam]  Get:7 http://archive.raspbian.org/raspbian stretch/firmware armhf Packages [1201 B]
[balena-cam]  Get:8 http://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]
[balena-cam]  Ign:5 http://deb.debian.org/debian jessie Release.gpg
[balena-cam]  Get:9 http://archive.raspbian.org/raspbian stretch/contrib armhf Packages [66.2 kB]
[balena-cam]  Get:10 http://deb.debian.org/debian jessie/main armhf Packages [8898 kB]
[balena-cam]  Get:11 http://archive.raspbian.org/raspbian stretch/rpi armhf Packages [1299 B]
[balena-cam]  Get:12 http://archive.raspbian.org/raspbian stretch/non-free armhf Packages [116 kB]
[balena-cam]  Get:13 http://archive.raspbian.org/raspbian stretch/main armhf Packages [16.4 MB]
[balena-cam]  Fetched 25.9 MB in 3s (6565 kB/s)
[balena-cam]  Reading package lists...
[balena-cam]  W: GPG error: http://deb.debian.org/debian jessie Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY CBF8D6FD518E17E1
[balena-cam]  W: The repository 'http://deb.debian.org/debian jessie Release' is not signed.
[balena-cam]  
[balena-cam]  Reading package lists...
[balena-cam]  Building dependency tree...
[balena-cam]  Reading state information...
[balena-cam]  The following additional packages will be installed:
[balena-cam]    dh-python fontconfig fontconfig-config fonts-dejavu-core gir1.2-atk-1.0
[balena-cam]    gir1.2-freedesktop gir1.2-gdkpixbuf-2.0 gir1.2-glib-2.0 gir1.2-gtk-2.0
[balena-cam]    gir1.2-pango-1.0 gnome-icon-theme gtk-update-icon-cache hicolor-icon-theme
[balena-cam]    icu-devtools iso-codes libasan3 libasound2 libasound2-data libass5
[balena-cam]    libasyncns0 libatk1.0-0 libatk1.0-data libatk1.0-dev libatomic1 libaudio2
[balena-cam]    libavahi-client3 libavahi-common-data libavahi-common3 libavc1394-0
[balena-cam]    libavcodec57 libavdevice57 libavfilter6 libavformat57 libavresample-dev
[balena-cam]    libavresample3 libavutil-dev libavutil55 libblas-common libbluray2 libbs2b0
[balena-cam]    libbsd0 libc-dev-bin libc6-dev libcaca0 libcairo-gobject2
[balena-cam]    libcairo-script-interpreter2 libcairo2 libcairo2-dev libcap2-bin
[balena-cam]    libcdio-cdda1 libcdio-paranoia1 libcdio13 libchromaprint1 libcroco3 libcups2
[balena-cam]    libcv-dev libcvaux-dev libdatrie1 libdc1394-22 libdc1394-22-dev
[balena-cam]    libdouble-conversion1 libdpkg-perl libebur128-1 libegl1-mesa libexpat1
[balena-cam]    libexpat1-dev libfftw3-double3 libflac8 libflite1 libfontconfig1
[balena-cam]    libfontconfig1-dev libfreetype6 libfreetype6-dev libfribidi0 libgbm1
[balena-cam]    libgcc-6-dev libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgdk-pixbuf2.0-dev
[balena-cam]    libgfortran3 libgirepository-1.0-1 libgl1-mesa-glx libglapi-mesa
[balena-cam]    libglib2.0-0 libglib2.0-bin libglib2.0-data libglib2.0-dev libgme0 libgomp1
[balena-cam]    libgraphite2-3 libgraphite2-dev libgsm1 libgstreamer-plugins-base1.0-0
[balena-cam]    libgstreamer1.0-0 libgtk2.0-0 libgtk2.0-common libgtk2.0-dev libharfbuzz-dev
[balena-cam]    libharfbuzz-gobject0 libharfbuzz-icu0 libharfbuzz0b libhighgui-dev
[balena-cam]    libice-dev libice6 libicu-dev libicu57 libiec61883-0 libilmbase-dev
[balena-cam]    libjack-jackd2-0 libjasper1 libjbig-dev libjbig0 libjpeg-dev libjpeg62-turbo
[balena-cam]    libjpeg62-turbo-dev liblcms2-2 liblzma-dev liblzo2-2 libmng1 libmp3lame0
[balena-cam]    libmpdec2 libmpg123-0 libogg0 libopenal-data libopenal1
[balena-cam]    libopencv-calib3d-dev libopencv-calib3d2.4v5 libopencv-contrib-dev
[balena-cam]    libopencv-contrib2.4v5 libopencv-core-dev libopencv-core2.4v5
[... 152 lines cut ...]
line with DEBUG=1, i.e.:  DEBUG=1 balena ...
If you need help, don't hesitate in contacting us at:

  Forums: https://forums.balena.io
  GitHub: https://github.com/balena-io/balena-cli/issues/new

```

</p>
</details>

This error can be easily be fixed by adding `--allow-unauthenticated` to the `apt-get install -yq` in the `Dockerfile.template` at line 6.

After that I encounter another issue that I am unable to solve, checkout the log (unrelevant bits have been removed.

```
[balena-cam]  ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
[balena-cam]      opencv-python==3.4.4.19 from https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.4.19-cp35-cp35m-linux_armv6l.whl#sha256=329d9d9fdd62b93d44a485aeaab4602c6f5b8555ea8bcc7dbcdc62c90cfe2c3f:
[balena-cam]          Expected sha256 329d9d9fdd62b93d44a485aeaab4602c6f5b8555ea8bcc7dbcdc62c90cfe2c3f
[balena-cam]               Got        3a32f926e1b319f5813ef8574b7dd3ac427768786ef0efc4a5256eb1a9b8bbe5
[balena-cam]  
[balena-cam]  Removing intermediate container f74fea06ff9b
[Info]        Uploading images
[balena-cam]  The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1
```

mbalamat, 2019-08-05:
@idoodler I looked into it, this is an issue with piwheels, I am working on a fix. Thanks for the feedback.

mbalamat, 2019-08-05:
@idoodler Looking into it

mbalamat, 2019-08-05:
@idoodler can you try this branch https://github.com/balena-io-playground/balena-cam/tree/fix_hash_mismatch

idoodler, 2019-08-05:
@mbalamat I can push the project, but the service keeps restarting on my Raspberry Pi Zero W

mbalamat, 2019-08-06:
@idoodler I located the error to `opencv-python` throwing `illegal instruction error` I've raised an issue on piwheels.

idoodler, 2019-08-06:
@mbalamat Nice, I am curious to know how you debugged it in Balena-Cloud.

idoodler, 2019-08-06:
Just for context, here is the linked issue: https://github.com/piwheels/packages/issues/43

mbalamat, 2019-08-06:
@idoodler so on balenaCloud what I did was to set the entrypoint to sleep and then attached to the balenaCam container and did the things I mention on the issue above.

mbalamat, 2019-11-09:
@idoodler can you try the branch on this PR https://github.com/balena-io-playground/balena-cam/pull/46

idoodler, 2019-11-10:
@mbalamat I now can push to Balena, but the project fails to run.

```
10.11.19 16:38:24 (+0100)  balena-cam  ImportError: numpy.core.multiarray failed to import
10.11.19 16:38:24 (+0100)  balena-cam  Traceback (most recent call last):
10.11.19 16:38:24 (+0100)  balena-cam    File "/usr/src/app/server.py", line 1, in <module>
10.11.19 16:38:24 (+0100)  balena-cam      import asyncio, json, os, cv2, platform, sys
10.11.19 16:38:24 (+0100)  balena-cam    File "/usr/local/lib/python3.5/site-packages/cv2/__init__.py", line 3, in <module>
10.11.19 16:38:24 (+0100)  balena-cam      from .cv2 import *
10.11.19 16:38:24 (+0100)  balena-cam  ImportError: numpy.core.multiarray failed to import
10.11.19 16:38:44 (+0100)  balena-cam  ImportError: numpy.core.multiarray failed to import
10.11.19 16:38:44 (+0100)  balena-cam  Traceback (most recent call last):
10.11.19 16:38:44 (+0100)  balena-cam    File "/usr/src/app/server.py", line 1, in <module>
10.11.19 16:38:44 (+0100)  balena-cam      import asyncio, json, os, cv2, platform, sys
10.11.19 16:38:44 (+0100)  balena-cam    File "/usr/local/lib/python3.5/site-packages/cv2/__init__.py", line 3, in <module>
10.11.19 16:38:44 (+0100)  balena-cam      from .cv2 import *
10.11.19 16:38:44 (+0100)  balena-cam  ImportError: numpy.core.multiarray failed to import
10.11.19 16:38:48 (+0100) Service exited 'balena-cam sha256:d0fc838918c4660a70127437850c7b8486405fe503fd8cada8029960c6cdb125'
10.11.19 16:38:53 (+0100) Restarting service 'balena-cam sha256:d0fc838918c4660a70127437850c7b8486405fe503fd8cada8029960c6cdb125'
```

idoodler, 2019-11-10:
@mbalamat If you don't have access to a Raspberry Pi Zero W you can send me the image, I will burn it to  an SD card, so you can debug it on a real device.

mbalamat, 2019-11-10:
@idoodler unboxing one now and setting it up to debug this...

mbalamat, 2019-11-10:
@idoodler my pi zero is probably fried can you add me as a developer at your balena cloud app that has a pi zero device with camera? You do this from the dashboard >> Members tab on the left

idoodler, 2019-11-11:
@mbalamat Sure, I just added you. My internet is currently terrible due to some provider issues, so fetching the project may be slow.

mbalamat, 2019-11-11:
@idoodler thanks I made some progress I'll check it later in the day. Everything build and imports correctly now. But there is an issue which camera device to use.

mbalamat, 2019-11-11:
@idoodler can you use this branch https://github.com/balena-io-playground/balena-cam/tree/pinned-base-image and try to change the capture device here https://github.com/balena-io-playground/balena-cam/blob/ddf6acd6ef3c88dd9e4886a612dd3de6318c4507/balena-cam/app/server.py#L10 also try to unplug/plug the camera on the device because I see 3 `/dev/video*`. For docs reference https://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_gui/py_video_display/py_video_display.html

idoodler, 2019-11-11:
@mbalamat Thx for your help. After reconnecting the Camera I successfully can fetch the MJPEG stream, however WebRTC is not working. Thats not a big deal, because I only need MJPEG. I guess the CPU is a bit to tiny.

Also, the device appears to be offline in balena cloud when starting a stream. So the fix was to update to the latest balena image + applying the fixes you made to this project?

idoodler, 2019-11-11:
@mbalamat Update, I just remembered about a WebRTC project I worked on back then. Firefox is working correctly, Chrome and Safari won't due to missing HTTPS connection to the Raspberry.

Do you still need to be a member of my project?

mbalamat, 2019-11-12:
@idoodler which project are you referring to? So just to clarify you just unplugged/plugged the camera and the everything works on the pi zero only when you use firefox?

idoodler, 2019-11-12:
@mbalamat Yes, I removed the Cameras Flex cable from the Raspberry Pi Zero W and plugged it back in again. That fixed the error with the missing Camera Device. I then wanted to view the camera stream via the local IP. Both Chrome and Safari Desktop won't show an image. Safari Mobile on iOS 13 (Only tried it externaly tho) and Firefox Desktop shows a camera stream.

The issue refered here is solved. Uploading to BalenaCloud works again. I will investigate in the Browser difference when I am at home again.

closed 2019-11-12 by idoodler

mbalamat, 2019-11-12:
@idoodler thanks for your help!

mbalamat, 2019-11-12:
@idoodler btw you can use the public url in order to use the app from the internet. And it also has https through balena's proxy. https://www.balena.io/docs/learn/manage/actions/#enable-public-device-url