balena-labs-projects/balena-cam #48: No matching distribution found for async-timeout

aweussom, 2019-11-15:
Tried to push BalenaCam to a Raspberry PI Zero W today. 

Got around the "missing GPG" error by changing line 6 to:
apt-get install -yq --allow-unauthenticated 
...to Dockerfile.template

But this one about Python has me stumped.

aweussom, 2019-11-15:
[main]     Step 8/10 : RUN pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple
[main]      ---> Running in 1878c35a3fed
[main]     Looking in indexes: https://www.piwheels.org/simple, https://www.piwheels.org/simple
[main]     Collecting aiohttp
[main]       Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-3.6.2-py3-none-any.whl (441kB)
[main]     Collecting aiohttp_basicauth==0.1.3
[main]       Downloading https://www.piwheels.org/simple/aiohttp-basicauth/aiohttp_basicauth-0.1.3-py3-none-any.whl
[main]     Collecting aiortc==0.9.11
[main]       Downloading https://www.piwheels.org/simple/aiortc/aiortc-0.9.11-cp35-cp35m-linux_armv6l.whl (112kB)
[main]     Collecting opencv-python==3.4.4.19
[main]       Downloading https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.4.19-cp35-cp35m-linux_armv6l.whl (8.7MB)
[main]     Collecting typing-extensions>=3.6.5; python_version < "3.7"
[main]       Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-3.7.4.1-py3-none-any.whl
[main]     Collecting idna-ssl>=1.0; python_version < "3.7"
[main]       Downloading https://www.piwheels.org/simple/idna-ssl/idna_ssl-1.1.0-py3-none-any.whl
[main]     Collecting attrs>=17.3.0
[main]       Downloading https://www.piwheels.org/simple/attrs/attrs-19.3.0-py2.py3-none-any.whl
[main]     Collecting multidict<5.0,>=4.5
[main]       Downloading https://www.piwheels.org/simple/multidict/multidict-4.5.2-cp35-cp35m-linux_armv6l.whl (301kB)
[main]     Collecting yarl<2.0,>=1.0
[main]       Downloading https://www.piwheels.org/simple/yarl/yarl-1.3.0-cp35-cp35m-linux_armv6l.whl (227kB)
[main]     Collecting chardet<4.0,>=2.0
[main]       Downloading https://www.piwheels.org/simple/chardet/chardet-3.0.4-py2.py3-none-any.whl (133kB)
[main]     ERROR: Could not find a version that satisfies the requirement async-timeout<4.0,>=3.0 (from aiohttp) (from versions: 4.0.0a0, 4.0.0a1, 4.0.0a2)
[main]
[main]     ERROR: No matching distribution found for async-timeout<4.0,>=3.0 (from aiohttp)
[main]
[main]     Removing intermediate container 1878c35a3fed
[main]     The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1
[Info]     Uploading images
[Success]  Successfully uploaded images
[Error]    Some services failed to build:
[Error]      Service: main
[Error]        Error: The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1
[Info]     Built on arm01
[Error]    Not deploying release.

mbalamat, 2019-11-15:
@aweussom there's this https://github.com/balena-io-playground/balena-cam/tree/pinned-base-image branch that I haven't merge yet. It's from this PR https://github.com/balena-io-playground/balena-cam/pull/46

aweussom, 2019-11-15:
That worked, after a couple of retries because of timeouts. Thanks!
I broke the camera-connector on my RPI Zero W - will test when I can get a new one.

mbalamat, 2019-11-17:
closing this if something comes up feel free to re-open it

closed 2019-11-17 by mbalamat