balena-labs-projects/balena-cam #50: Weird error - "/usr/lib/arm-linux-gnueabihf/libxml2.so.2: unsupported version 0 of Verdef record"

robot-army, 2019-12-12:
Hihi, 

I'm trying to run this on a Pi Zero W, but it gets stuck in a service restart loop. 
```
11.12.19 21:38:43 (-0800) Supervisor starting
11.12.19 21:39:01 (-0800) Applying configuration change {"SUPERVISOR_POLL_INTERVAL":"900000","SUPERVISOR_DELTA":"1","SUPERVISOR_DELTA_VERSION":"3"}
11.12.19 21:39:01 (-0800) Applying boot config: {"gpu_mem_1024":"448","gpu_mem_256":"192","gpu_mem_512":"256","start_x":"1","avoid_warnings":"1","disable_splash":"1","dtparam":["i2c_arm=on","spi=on","audio=on"],"gpu_mem":"16"}
11.12.19 21:39:02 (-0800) Applied boot config: {"gpu_mem_1024":"448","gpu_mem_256":"192","gpu_mem_512":"256","start_x":"1","avoid_warnings":"1","disable_splash":"1","dtparam":["i2c_arm=on","spi=on","audio=on"],"gpu_mem":"16"}
11.12.19 21:39:02 (-0800) Applied configuration change {"SUPERVISOR_POLL_INTERVAL":"900000","SUPERVISOR_DELTA":"1","SUPERVISOR_DELTA_VERSION":"3"}
11.12.19 21:39:04 (-0800) Rebooting
11.12.19 21:41:35 (-0800) Supervisor starting
11.12.19 21:41:44 (-0800) Creating network 'default'
11.12.19 21:48:39 (-0800) Downloading image 'registry2.balena-cloud.com/v2/898cd2b6d672d4b05645347dbbf606d0@sha256:136ed155eefef960c3feddbb8f76fe49803ece0a52d064d73adcb642acdae167'
11.12.19 22:12:43 (-0800) Downloaded image 'registry2.balena-cloud.com/v2/898cd2b6d672d4b05645347dbbf606d0@sha256:136ed155eefef960c3feddbb8f76fe49803ece0a52d064d73adcb642acdae167'
11.12.19 22:12:46 (-0800) Installing service 'balena-cam sha256:f0c7276b7116b308f1dba9a7cde3318e45fb3ea7d83f5c690a2ff7ed36bf7c52'
11.12.19 22:13:08 (-0800) Installed service 'balena-cam sha256:f0c7276b7116b308f1dba9a7cde3318e45fb3ea7d83f5c690a2ff7ed36bf7c52'
11.12.19 22:13:08 (-0800) Starting service 'balena-cam sha256:f0c7276b7116b308f1dba9a7cde3318e45fb3ea7d83f5c690a2ff7ed36bf7c52'
11.12.19 22:13:28 (-0800) Started service 'balena-cam sha256:f0c7276b7116b308f1dba9a7cde3318e45fb3ea7d83f5c690a2ff7ed36bf7c52'
11.12.19 22:13:36 (-0800) <balena-cam> Traceback (most recent call last):
11.12.19 22:13:36 (-0800) <balena-cam>   File "/usr/src/app/server.py", line 1, in <module>
11.12.19 22:13:36 (-0800) <balena-cam>     import asyncio, json, os, cv2, platform, sys
11.12.19 22:13:36 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/cv2/__init__.py", line 3, in <module>
11.12.19 22:13:36 (-0800) <balena-cam>     from .cv2 import *
11.12.19 22:13:36 (-0800) <balena-cam> ImportError: /usr/lib/arm-linux-gnueabihf/libxml2.so.2: unsupported version 0 of Verdef record```

robot-army, 2019-12-12:
It's only got a 4 GB SD card, but it doesn't look full to me: 

```
root@0827b25:~# df -h
Filesystem                         Size  Used Avail Use% Mounted on
devtmpfs                           114M     0  114M   0% /dev
/dev/disk/by-partuuid/9af45c83-02  300M  219M   61M  79% /mnt/sysroot/active
/dev/disk/by-label/resin-state      19M  216K   17M   2% /mnt/state
none                               300M  219M   61M  79% /
tmpfs                              122M  108K  122M   1% /dev/shm
tmpfs                              122M  4.8M  117M   4% /run
tmpfs                              122M     0  122M   0% /sys/fs/cgroup
tmpfs                              122M     0  122M   0% /tmp
/dev/mmcblk0p1                      40M  8.2M   32M  21% /mnt/boot
tmpfs                              122M   28K  122M   1% /var/volatile
/dev/mmcblk0p6                     2.9G  958M  1.8G  36% /mnt/data
/dev/mmcblk0p3                     300M  2.1M  278M   1% /mnt/sysroot/inactive
```

mbalamat, 2019-12-16:
@robot-army I haven't encountered something like this. Can you use a different SD card and retry?

robot-army, 2019-12-17:
Yep, using a different SD card fixed this error. Thanks!

robot-army, 2019-12-17:
df -h now reads:

```
root@8d5b3ae:~# df -h
Filesystem                         Size  Used Avail Use% Mounted on
devtmpfs                           114M     0  114M   0% /dev
/dev/disk/by-partuuid/9af45c83-02  300M  219M   61M  79% /mnt/sysroot/active
/dev/disk/by-label/resin-state      19M  217K   17M   2% /mnt/state
none                               300M  219M   61M  79% /
tmpfs                              122M  212K  121M   1% /dev/shm
tmpfs                              122M  3.8M  118M   4% /run
tmpfs                              122M     0  122M   0% /sys/fs/cgroup
tmpfs                              122M     0  122M   0% /tmp
/dev/mmcblk0p1                      40M  8.2M   32M  21% /mnt/boot
tmpfs                              122M   52K  122M   1% /var/volatile
/dev/mmcblk0p6                     115G  1.2G  108G   2% /mnt/data
/dev/mmcblk0p3                     300M  2.1M  278M   1% /mnt/sysroot/inactive```

closed 2019-12-17 by robot-army