balena-labs-projects/balena-cam #55: Error on fresh build: VIDEOIO ERROR: V4L: can't open camera by index 0

robot-army, 2019-12-17:
Hi, 

After installing a larger SD card (a big jump... 4G-128G) the previous strange error I was having stopped (see issue 

16.12.19 21:26:09 (-0800)  balena-cam  VIDEOIO ERROR: V4L: can't open camera by index 0
16.12.19 21:26:09 (-0800)  balena-cam  Failed to open default camera. Exiting...

I changed the index of the video device to -1 and got the error 
"VIDEOIO ERROR: V4L: can't find camera device" instead. 

Issuing the command 'modprobe bcm2835-v4l2' seems to fix it, but it doesn't persist after a reboot. It seems a recent commit removed that check?

Now, it only fixed THAT problem. Now the stream runs for a second, maybe a few frames, then the whole thing stops. I suspect it's a bit much for the pi zero to handle, but if anyone is having luck with that hardware, it must be something else. 

I'm trying now to change    'await asyncio.sleep'... to 5 seconds in server.py. I reckon that should throttle it enough to let the pi zero's tiny brain catch up.

mbalamat, 2019-12-18:
@robot-army which balenaOS version are you using? Also did the throttling work for you?

robot-army, 2019-12-19:
Was using BalenaOS version balenaOS 2.32.0+rev1 . Throttling in that way didn't work, I made a new issue for that. It was one of the GPU memory settings.

mbalamat, 2019-12-19:
@robot-army do yo mind updating to the latest version

mbalamat, 2019-12-19:
@robot-army you can do this by clicking on your device > Actions (it's on the left side of the dashboard) > BalenaOS Update > click update BalenaOS

robot-army, 2019-12-21:
Hmmmm.... I get this when trying to update?

[image]

robot-army, 2019-12-21:
On my Pi Zero, it's the latest version, it seems!

mbalamat, 2019-12-30:
@robot-army some versions of the OS got retracted recently because of an issue. The latest recommended is 2.32 for the RPi zero. I am currently testing to see if balenaCam works on this version. In general the last time I added a commit everything was working for the 2.4x versions of the OS.

mbalamat, 2019-12-31:
hey @robot-army I pinpointed the issue, you can check my response on balena forums https://forums.balena.io/t/balena-cam-raspberry-p-v4l-cant-open-camera/52485/19

robot-army, 2020-01-03:
Ahh OK cool, looks like the right solution for now is to wait a couple weeks for v2.4x for the Pi Zero. Should we keep this issue open until then?

mbalamat, 2020-01-03:
@robot-army yes let's keep it open and confirm the results when the v2.4x for the pi zero rolls out.

webtop, 2020-01-15:
I'm running v2.38 and was having the same issue. I uncommented that line you mention in the post, nothing else, and it's working as far as it loads, but I just get a blank view. I see the application framework, but the camera feed is just blank.

Are these issues related or should I start a new thread?

mbalamat, 2020-01-15:
@webtop yes as you are running 2.38 there is an issue with the camera libs that will be fixed in balenaOS version 2.4x when it comes out for the piZero.

webtop, 2020-01-15:
Thx Marios.

mbalamat, 2020-03-02:
@robot-army @webtop you can try the master branch now and re-open this (or other) if any issues occur.

closed 2020-03-02 by mbalamat

webtop, 2020-03-19:
Hey Marios,

I just pulled down the latest master and rebuilt. No issues in the build process, but once the app started, I get this issue:

> [Logs]    [3/18/2020, 6:42:34 PM] Started service 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:42:39 PM] [balena-cam] Video device is not ready
> [Logs]    [3/18/2020, 6:42:43 PM] Service exited 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:42:45 PM] Restarting service 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:42:47 PM] [balena-cam] Trying to load bcm2835-v4l2 driver...
> [Logs]    [3/18/2020, 6:42:51 PM] Service exited 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:42:52 PM] Restarting service 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:42:55 PM] [balena-cam] Video device is not ready
> [Logs]    [3/18/2020, 6:42:58 PM] Service exited 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'
> [Logs]    [3/18/2020, 6:43:00 PM] Restarting service 'balena-cam sha256:a01d0483066dcbbf558f7400897337219a44d1dd38004a2d8c008c03764fbf7d'

I'm doing this all locally with the CLI, but just realized I'm still running 2.38...how do I update to 2.4 from local CLI? I can't find any info, and "balena device os-update" expect a UUID not a host IP.

Thx :)

webtop, 2020-03-19:
FYI: I ssh'd into the device, and tried:

> balena-engine engine update --version v2.46.1+rev3.dev
but got 
> unable to access local containerd: failed to dial "/run/containerd/containerd.sock": context deadline exceeded

mbalamat, 2020-03-24:
@webtop why not just use the baelnaDashboard to update?

mbalamat, 2020-03-24:
And if you want to use the cli you can do what you say and acquire the uuid from `balena scan`.

webtop, 2020-03-24:
Hey! It's all good now. I have inadvertently tried to flash an incorrect device image, which then bricked the device. I ended up just reflashing the device with the latest update.

My devices are local dev devices, not connected to the dashboard. Are you saying, just perform an attach, then an update, then a detach? If so....good idea :)

mbalamat, 2020-03-25:
@webtop is your device provisioned on balenaCloud? If yes, then even though you have a dev flavour of the balenaOS and you are using local mode you can just, as you say, disable local mode for a while, perform the update and re-enable it. In general with a device running a dev version of balenaOS, provisioned on balenaCloud, you can do the same things as with a device running a production version of the OS. Plus with dev balenaOS images you can do extra stuff like local mode and local push, so you can have fast development cycles/iterations.