balena-labs-projects/balena-cam #56: Feed stops after some short time (frames)

robot-army, 2019-12-17:


robot-army, 2019-12-17:
```
16.12.19 22:23:45 (-0800) <balena-cam> Exception in callback _SelectorTransport._call_connection_lost(None)
16.12.19 22:23:45 (-0800) <balena-cam> handle: <Handle _SelectorTransport._call_connection_lost(None)>
16.12.19 22:23:45 (-0800) <balena-cam> Traceback (most recent call last):
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
16.12.19 22:23:45 (-0800) <balena-cam>     self._callback(*self._args)
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/selector_events.py", line 662, in _call_connection_lost
16.12.19 22:23:45 (-0800) <balena-cam>     self._protocol.connection_lost(exc)
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/ice.py", line 161, in connection_lost
16.12.19 22:23:45 (-0800) <balena-cam>     self.receiver.data_received(None, self.local_candidate.component)
16.12.19 22:23:45 (-0800) <balena-cam> AttributeError: 'NoneType' object has no attribute 'component'
16.12.19 22:23:45 (-0800) <balena-cam> Exception in callback _SelectorTransport._call_connection_lost(None)
16.12.19 22:23:45 (-0800) <balena-cam> handle: <Handle _SelectorTransport._call_connection_lost(None)>
16.12.19 22:23:45 (-0800) <balena-cam> Traceback (most recent call last):
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
16.12.19 22:23:45 (-0800) <balena-cam>     self._callback(*self._args)
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/selector_events.py", line 662, in _call_connection_lost
16.12.19 22:23:45 (-0800) <balena-cam>     self._protocol.connection_lost(exc)
16.12.19 22:23:45 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/ice.py", line 161, in connection_lost
16.12.19 22:23:45 (-0800) <balena-cam>     self.receiver.data_received(None, self.local_candidate.component)
16.12.19 22:23:45 (-0800) <balena-cam> AttributeError: 'NoneType' object has no attribute 'component'
```

robot-army, 2019-12-17:
Output from top seems like kswapd0 takes over CPU usage right when the thing fails. I've tried changing the GPU memory settings, let's see how that goes.

robot-army, 2019-12-17:
Yep, by far the longest run I've had yet (and it's still running as I post this comment) has come by overriding device configuration ```BALENA_HOST_CONFIG_gpu_mem_512``` value from 256 to 128 (from here: https://www.balena.io/docs/learn/develop/hardware/i2c-and-spi/#raspberry-pi-camera-module )

robot-army, 2019-12-17:
Some noise in the logs and a broken feed that seemed to work on refresh: 

```17.12.19 01:55:47 (-0800) <balena-cam> Exception in callback Transaction.__retry()
17.12.19 01:55:47 (-0800) <balena-cam> handle: <TimerHandle when=1585.787987345 Transaction.__retry()>
17.12.19 01:55:47 (-0800) <balena-cam> Traceback (most recent call last):
17.12.19 01:55:47 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
17.12.19 01:55:47 (-0800) <balena-cam>     self._callback(*self._args)
17.12.19 01:55:47 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/stun.py", line 272, in __retry
17.12.19 01:55:47 (-0800) <balena-cam>     self.__future.set_exception(exceptions.TransactionTimeout())
17.12.19 01:55:47 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/futures.py", line 360, in set_exception
17.12.19 01:55:47 (-0800) <balena-cam>     raise InvalidStateError('{}: {!r}'.format(self._state, self))
17.12.19 01:55:47 (-0800) <balena-cam> asyncio.futures.InvalidStateError: FINISHED: <Future finished result=(Message(messa...\x97"ZM\x1f]'), ('192.168.0.26', 58452))>
17.12.19 01:57:15 (-0800) <balena-cam> Exception in callback Transaction.__retry()
17.12.19 01:57:15 (-0800) <balena-cam> handle: <TimerHandle when=1675.406446503 Transaction.__retry()>
17.12.19 01:57:15 (-0800) <balena-cam> Traceback (most recent call last):
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
17.12.19 01:57:15 (-0800) <balena-cam>     self._callback(*self._args)
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/stun.py", line 272, in __retry
17.12.19 01:57:15 (-0800) <balena-cam>     self.__future.set_exception(exceptions.TransactionTimeout())
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/futures.py", line 360, in set_exception
17.12.19 01:57:15 (-0800) <balena-cam>     raise InvalidStateError('{}: {!r}'.format(self._state, self))
17.12.19 01:57:15 (-0800) <balena-cam> asyncio.futures.InvalidStateError: FINISHED: <Future finished exception=TransactionFailed(Message(message_method=Method.BINDING, message_class=Class.ERROR, transaction_id=b'\x9a\xe6\xa6\xc3\xfd\xf3\xd8j\x86\xd7OE'),)>
17.12.19 01:57:15 (-0800) <balena-cam> Exception in callback Transaction.__retry()
17.12.19 01:57:15 (-0800) <balena-cam> handle: <TimerHandle when=1676.390663887 Transaction.__retry()>
17.12.19 01:57:15 (-0800) <balena-cam> Traceback (most recent call last):
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
17.12.19 01:57:15 (-0800) <balena-cam>     self._callback(*self._args)
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/stun.py", line 272, in __retry
17.12.19 01:57:15 (-0800) <balena-cam>     self.__future.set_exception(exceptions.TransactionTimeout())
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/futures.py", line 360, in set_exception
17.12.19 01:57:15 (-0800) <balena-cam>     raise InvalidStateError('{}: {!r}'.format(self._state, self))
17.12.19 01:57:15 (-0800) <balena-cam> asyncio.futures.InvalidStateError: FINISHED: <Future finished exception=TransactionFailed(Message(message_method=Method.BINDING, message_class=Class.ERROR, transaction_id=b'?\xf1`\x96;\xcb*\xb0(\xa4kG'),)>
17.12.19 01:57:15 (-0800) <balena-cam> Exception in callback Transaction.__retry()
17.12.19 01:57:15 (-0800) <balena-cam> handle: <TimerHandle when=1676.409545741 Transaction.__retry()>
17.12.19 01:57:15 (-0800) <balena-cam> Traceback (most recent call last):
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
17.12.19 01:57:15 (-0800) <balena-cam>     self._callback(*self._args)
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/stun.py", line 272, in __retry
17.12.19 01:57:15 (-0800) <balena-cam>     self.__future.set_exception(exceptions.TransactionTimeout())
17.12.19 01:57:15 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/futures.py", line 360, in set_exception
17.12.19 01:57:15 (-0800) <balena-cam>     raise InvalidStateError('{}: {!r}'.format(self._state, self))
17.12.19 01:57:15 (-0800) <balena-cam> asyncio.futures.InvalidStateError: FINISHED: <Future finished exception=TransactionFailed(Message(message_method=Method.BINDING, message_class=Class.ERROR, transaction_id=b'\xc5\xec\xb5x\xe05\xe4\x8d\xde~\xc7r'),)>
17.12.19 02:00:38 (-0800) <balena-cam> Exception in callback Transaction.__retry()
17.12.19 02:00:38 (-0800) <balena-cam> handle: <TimerHandle when=1874.514632089 Transaction.__retry()>
17.12.19 02:00:38 (-0800) <balena-cam> Traceback (most recent call last):
17.12.19 02:00:38 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
17.12.19 02:00:38 (-0800) <balena-cam>     self._callback(*self._args)
17.12.19 02:00:38 (-0800) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/stun.py", line 272, in __retry
17.12.19 02:00:38 (-0800) <balena-cam>     self.__future.set_exception(exceptions.TransactionTimeout())
17.12.19 02:00:38 (-0800) <balena-cam>   File "/usr/lib/python3.5/asyncio/futures.py", line 360, in set_exception
17.12.19 02:00:38 (-0800) <balena-cam>     raise InvalidStateError('{}: {!r}'.format(self._state, self))
17.12.19 02:00:40 (-0800) <balena-cam> asyncio.futures.InvalidStateError: FINISHED: <Future finished result=(Message(messa...d1m\xb8\x12>'), ('192.168.0.26', 62341))>```

Also, this is after uncommenting the line commented here
```if __name__ == '__main__':
    #checkDeviceReadiness() ```

mbalamat, 2019-12-19:
@robot-army since you don't have this problem anymore do you mind closing this? Also I'd recommend always using the latest available version of balenaOS.

robot-army, 2019-12-20:
I think this needs something for the README to fix the GPU memory sizes, but I'll try again with a new version if I get a moment and see if that fixes it. Thanks for your help!

mbalamat, 2020-03-02:
@robot-army I am going to close this, if you still have the issue feel free to re-open it.

closed 2020-03-02 by mbalamat