balena-labs-projects/balena-cam #57: Trouble building (in local mode) on a Raspberry Pi 4

robot-army, 2019-12-19:
Finally got this working on a Pi Zero W by changing the GPU memory settings. 

Moving to a Raspberry Pi 4, with local deployment, and after balena push x.x.x.x
```some services failed to build:
	balena-cam: manifest for balenalib/raspberrypi4-64-debian:stretch-20190612 not found```

Changed first line of Dockerfile.template to 
```FROM balenalib/%%BALENA_MACHINE_NAME%%-debian:stretch```

Now got the error 
```[Build]   [balena-cam] E: Unable to locate package libjasper-dev```

Commented out the line for that in the same dockerfile.template, but I'm sure it wouldn't be there if it wasn't needed, so I'm sure that'll cause problems down the line. 

Now getting the error: 
```[Build]   [balena-cam] Collecting aiohttp
[Build]   [balena-cam]   Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-3.6.2-py3-none-any.whl (441kB)
[Build]   [balena-cam] Collecting aiohttp_basicauth==0.1.3
[Build]   [balena-cam]   Downloading https://www.piwheels.org/simple/aiohttp-basicauth/aiohttp_basicauth-0.1.3-py3-none-any.whl
[Build]   [balena-cam] ERROR: Could not find a version that satisfies the requirement aiortc==0.9.11 (from versions: none)
[Build]   
[Build]   [balena-cam] ERROR: No matching distribution found for aiortc==0.9.11
[Build]   
[Build]   [balena-cam] Removing intermediate container 0ba8bf495a40
Some services failed to build:
	balena-cam: The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1```

mbalamat, 2019-12-19:
@robot-army pi4 can run 32bit containers so if you create another dockerfile which is the same as the one here and name it `Dockerfile.amd64` and its first line be `FROM balenalib/raspberrypi3-debian:stretch-20190612` it should work. If you try it do you mind PRing the changes?

robot-army, 2019-12-20:
Sure, no sweat. I'll keep up to date here. Just one thing, do you mean Dockerfile.amd64? The AMD bit seems out of place on an ARM architecture (but I'm a hardware guy, so...I'm really just swinging a hammer at stuff for now!) 

Thanks for coming back to me. I'll shout up in other issues as well!

mbalamat, 2019-12-20:
@robot-army sorry man it was late when I responded it should be `Dockerfile.aarch64`

robot-army, 2019-12-21:
OK, I just reverted my other changes and tried that, and it's still coming with the error below. 

```
Ryans-MBP:balena-cam robotarmy$ balena push 192.168.0.43
[debug] original argv0="/usr/local/lib/balena-cli/bin/node" argv=[/usr/local/lib/balena-cli/bin/node,/usr/local/lib/balena-cli/bin/run,push,192.168.0.43] length=4
[debug] Using . as build source
[Debug]   Checking we can access device
[Debug]   Sending request to http://192.168.0.43:48484/ping
[Debug]   Checking device version: 10.2.2
[Info]    Starting build on device 192.168.0.43
[Debug]   Loading project...
[Debug]   Resolving project...
[Info]    Compose file detected
[Debug]   Creating project...
[Debug]   Tarring all non-ignored files...
[Debug]   Sending request to http://192.168.0.43:48484/v2/local/device-info
[Debug]   Found build tasks:
[Debug]       balena-cam: build [./balena-cam]
[Debug]   Resolving services with [raspberrypi4-64|aarch64]
[Debug]   Found project types:
[Debug]       balena-cam: Architecture-specific Dockerfile
[Debug]   Probing remote daemon for cache images
[Debug]   Using 15 on-device images for cache...
[Debug]   Starting builds...
[Build]   [balena-cam] Step 1/11 : FROM balenalib/raspberrypi3-debian:stretch-20190612
[Build]   [balena-cam]  ---> 9a857048cc34
[Build]   [balena-cam] Step 2/11 : RUN apt-get update &&   apt-get install -yq     python3     python3-dev     python3-pip     python3-setuptools     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     libsrtp2-dev     libopencv-dev     libatlas3-base     libatlas-base-dev     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*
[Build]   [balena-cam]  ---> Using cache
[Build]   [balena-cam]  ---> d33dc9a32d86
[Build]   [balena-cam] Step 3/11 : RUN pip3 install --upgrade pip
[Build]   [balena-cam]  ---> Running in 7d3f5a83bbab
[Build]   [balena-cam] Collecting pip
[Build]   [balena-cam]   Downloading https://files.pythonhosted.org/packages/00/b6/9cfa56b4081ad13874b0c6f96af8ce16cfbc1cb06bedf8e9164ce5551ec1/pip-19.3.1-py2.py3-none-any.whl (1.4MB)
[Build]   [balena-cam] Installing collected packages: pip
[Build]   [balena-cam]   Found existing installation: pip 9.0.1
[Build]   [balena-cam]     Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
[Build]   [balena-cam] Successfully installed pip-19.3.1
[Build]   [balena-cam] Removing intermediate container 7d3f5a83bbab
[Build]   [balena-cam]  ---> 15573ce840f5
[Build]   [balena-cam] Step 4/11 : RUN pip3 install async-timeout
[Build]   [balena-cam]  ---> Running in 55252b95dfe9
[Build]   [balena-cam] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[Build]   [balena-cam] Collecting async-timeout
[Build]   [balena-cam]   Downloading https://files.pythonhosted.org/packages/e1/1e/5a4441be21b0726c4464f3f23c8b19628372f606755a9d2e46c187e65ec4/async_timeout-3.0.1-py3-none-any.whl
[Build]   [balena-cam] Installing collected packages: async-timeout
[Build]   [balena-cam] Successfully installed async-timeout-3.0.1
[Build]   [balena-cam] Removing intermediate container 55252b95dfe9
[Build]   [balena-cam]  ---> be51a2e2a81c
[Build]   [balena-cam] Step 5/11 : RUN pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple
[Build]   [balena-cam]  ---> Running in b38292728616
[Build]   [balena-cam] Looking in indexes: https://www.piwheels.org/simple, https://www.piwheels.org/simple
[Build]   [balena-cam] Collecting aiohttp
[Build]   [balena-cam]   Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-3.6.2-py3-none-any.whl (441kB)
[Build]   [balena-cam] Collecting aiohttp_basicauth==0.1.3
[Build]   [balena-cam]   Downloading https://www.piwheels.org/simple/aiohttp-basicauth/aiohttp_basicauth-0.1.3-py3-none-any.whl
[Build]   [balena-cam] ERROR: Could not find a version that satisfies the requirement aiortc==0.9.11 (from versions: none)
[Build]   
[Build]   [balena-cam] ERROR: No matching distribution found for aiortc==0.9.11
[Build]   
[Build]   [balena-cam] Removing intermediate container b38292728616
Some services failed to build:
	balena-cam: The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1
```

mbalamat, 2019-12-23:
@robot-army can you PR the changes so I can test with my pi4?

robot-army, 2020-01-03:
OK Done, see https://github.com/balena-io-projects/balena-cam/pull/58

robot-army, 2020-01-03:
I'm not sure what configuration from Balena itself might be useful, let me know if anything.

schottsfired, 2020-02-12:
This command works:
`docker build -f Dockerfile.template -t balena-cam:latest .`

And this command fails with dependency resolution errors:
`balena push balena-cam`

So it seems like a Balena's `docker build` is different than a local `docker build` and something on the Balena side is causing the issue.

mbalamat, 2020-02-14:
@schottsfired is your device a pi4?

schottsfired, 2020-02-14:
@mbalamat I'm using a Pi3B+ with the `balenalib/raspberrypi3-debian:stretch-20190612` image.

ChadTaljaardt, 2020-02-19:
Getting the same issue on Pi 4

mbalamat, 2020-02-19:
@schottsfired I haven't tried to build it locally, in balena's builders it builds successfully, I tried it right now.

mbalamat, 2020-02-19:
@ChadTaljaardt in general we didn't implement any specific support for pi4. The way to get balena-cam running on pi4 is actually by building it for the pi3, because the 32bit(pi3) containers can run on 64bits(pi4). Probably an easy way to accomplish this is by making an application for pi3 in balena-cloud, push balena-cam there and then move your pi4 to this exact application.

mbalamat, 2020-02-19:
Also if you guys would like to build support for the 64bits we are open for PRs.

mbalamat, 2020-02-19:
Also that fallback to 32bits is implemented in this PR #58 you could also try that.

schottsfired, 2020-02-21:
The only difference I can see between #58 and the `master` branch is line #1 uses an image directly instead of a code reference. I'm doing the same thing, but with a newer base image:
`FROM balenalib/raspberrypi3-debian:stretch-20200207`

Still, the `balena push` command fails with the trace reported by @robot-army in https://github.com/balenalabs/balena-cam/issues/57#issuecomment-568156667. I upgraded the `balena` CLI to the latest version, but it didn't help.

Local `docker build` works fine:
```
Step 5/9 : RUN pip3 install aiohttp aiohttp_basicauth==0.1.3 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple
 ---> Running in 9a6d457ac8e2
Looking in indexes: https://www.piwheels.org/simple, https://www.piwheels.org/simple
Collecting aiohttp
  Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-3.6.2-py3-none-any.whl (441 kB)
Collecting aiohttp_basicauth==0.1.3
  Downloading https://www.piwheels.org/simple/aiohttp-basicauth/aiohttp_basicauth-0.1.3-py3-none-any.whl (7.0 kB)
Collecting aiortc==0.9.11
  Downloading https://www.piwheels.org/simple/aiortc/aiortc-0.9.11-cp35-cp35m-linux_armv7l.whl (112 kB)
...
```

 Is `balena push` pulling from a cache instead of doing a fresh `docker build`?

> in balena's builders it builds successfully, I tried it right now.

That's great, but I think it's still failing for the three other people in this thread.

mbalamat, 2020-02-21:
@schottsfired the #58 branch is not meant for you since you are using a pi3. Can you provide some more info about your setup? BalenaOS version mainly and if you have the 64bit version of the OS? Also the full output of `balena push <yourAppsName>`

mbalamat, 2020-02-21:
Also yes the builders use caching.

mbalamat, 2020-03-02:
@robot-army @schottsfired @ChadTaljaardt I am closing this. There was peer dependency issue, which I fixed, you can use the master branch (v0.1.0). If you encounter any issues feel free to re-open it, or better open another issue.

closed 2020-03-02 by mbalamat