balena-labs-projects/balena-cam #61: Invalid IP address issue

pallsopp, 2020-03-21:
Getting the following error. UI starts as expected and is functional, but the stream is not being fetched correctly.

I am using typical RPi camera board, connected to a Fin board.

> Video device is ready
> 
> #############################################################
> Authorization is disabled.
> Anyone can access your balenaCam, using the device's URL!
> Set the username and password environment variables
> to enable authorization.
> For more info visit:
> https://github.com/balena-io-playground/balena-cam
> #############################################################
> 
> ======== Running on http://0.0.0.0:80 ========
> (Press CTRL+C to quit)
> Task exception was never retrieved
> future: <Task finished coro=<RTCPeerConnection.__connect() done, defined at /usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py:554> exception=ValueError("'88803ef1-075a-4f4d-9954-9f4f4897632a.local' does not appear to be an IPv4 or IPv6 address",)>
> Traceback (most recent call last):
>   File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
>     result = coro.send(None)
>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py", line 559, in __connect
>     await iceTransport.start(self.__remoteIce[transceiver])
>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcicetransport.py", line 284, in start
>     await self._connection.connect()
>   File "/usr/local/lib/python3.5/dist-packages/aioice/ice.py", line 356, in connect
>     if (protocol.local_candidate.can_pair_with(remote_candidate) and
>   File "/usr/local/lib/python3.5/dist-packages/aioice/candidate.py", line 114, in can_pair_with
>     b = ipaddress.ip_address(other.host)
>   File "/usr/lib/python3.5/ipaddress.py", line 54, in ip_address
>     address)
> ValueError: '88803ef1-075a-4f4d-9954-9f4f4897632a.local' does not appear to be an IPv4 or IPv6 address

I'm planning on debugging the issue this evening as I have plenty of Python experience, but thought I'd post here in case this is a known issue.

My Fin is running in local mode, running containers, with the cam project running from one of the containers. Host OS is 2.46, so cam libs should all be fine.

Thx

mbalamat, 2020-03-24:
@pallsopp the error that pops out to me is this one `exception=ValueError("'88803ef1-075a-4f4d-9954-9f4f4897632a.local' does not appear to be an IPv4 or IPv6 address",)>` What browser are you using?

mbalamat, 2020-03-24:
I think if you disable the Local IP hiding for webrtc on your browser, it will work, I'll close this for now, feel free to re-open it, if it's something else.

closed 2020-03-24 by mbalamat

webtop, 2020-03-24:
Thanks Marios, that did the trick.

Anyone else who has this issue: if you're using Chrome, the flag is here:
chrome://flags/#enable-webrtc-hide-local-ips-with-mdns

mbalamat, 2020-03-25:
@webtop awesome! thanks for putting the instructions here. We are open for PRs if you are up adding this valuable info to our readme!

webtop, 2020-03-26:
@mbalamat Added PR 62.

Thanks :)