balena-labs-projects/balena-cam #64: libjasper-dev missing again

Esser420, 2020-03-28:
I saw that this has been an issue before and that there was a fix at that time. Now builds from master are failing again due to the libjasper-dev package missing.

mbalamat, 2020-04-01:
Hello @Esser420 I just tried to build it against a pi3 application and it works. Can you provide some more info about your setup?
Device type?
Camera module?
BalenaOS version?

Esser420, 2020-04-03:
Ok I had a friend recently got the build working on an rpi0, I'll check if I didn't do any mistake.

mbalamat, 2020-04-04:
@Esser420 you can paste here the output of the build. Especially the lines that say what base image was used, this might tell us something.

mbalamat, 2020-04-06:
closing for now, if you encounter the issue feel free to re-open this,

closed 2020-04-06 by mbalamat