balena-labs-projects/balena-cam #68: Capturing video from second camera on balenaFin

Hegazi, 2020-05-14:
I have enabled 'Dual camera mode' as per instructions for the balenaFin: https://www.balena.io/fin/1.1/docs/developers/#Dual-camera-mode 
I also changed the Fleet Configuration. I get the following output when switching cv2.VideoCapture(0) in servery.py to cv2.VideoCapture(1) to get the feed of the second camera.
Note: Works fine with main camera
```
 balena-cam  select timeout
 balena-cam  VIDIOC_DQBUF: Resource temporarily unavailable
 balena-cam  Failed to open default camera. Exiting...
 balena-cam  Video device is ready
 balena-cam  select timeout
 balena-cam  VIDIOC_DQBUF: Resource temporarily unavailable
 balena-cam  Failed to open default camera. Exiting...
Service exited 'balena-cam'
Restarting service 'balena-cam' 
```

mbalamat, 2020-05-15:
Hi @Hegazi can before running balena-cam can you check if there is a /dev/video1 device?

Hegazi, 2020-05-16:
Running 'ls /dev/' I get video0, video1, video10, video11, video12.

mbalamat, 2020-05-21:
@Hegazi sorry for the late reply. Can you try running `vcgencmd get_camera` to see if the camera's are recognised. After that can you try to run `raspistill -cs 1 -o image1.jpg` to see if the camera that is connected to the second CSI port works and takes a picture?

Hegazi, 2020-05-22:
@mbalamat I actually have posted on the forum right [here](https://forums.balena.io/t/set-state-of-dsi-cam1-switch/32111?u=hegazi): 

Running `vcgencmd get_camera` I get:

> supported=2 detected=2

Running `raspistill -cs 1 -o image1.jpg` I get:

> mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
> mmal: camera component couldn't be enabled
> mmal: main: Failed to create camera component
> mmal: Failed to run camera app. Please check for firmware updates

rahul-thakoor, 2020-05-22:
Have you also changed [this line](https://github.com/balenalabs/balena-cam/blob/master/balena-cam/app/server.py#L168) ?

mbalamat, 2020-05-27:
@Hegazi since we are discussing this on the forums I'll close this. Feel free to re-open it if you need to.

closed 2020-05-27 by mbalamat