balena-labs-projects/balena-cam #73: No video, probable WebRTC problem on Pi4

HughWarrington, 2020-06-26:
I cloned the repo and followed the instructions in README.md.

I am using a Raspberry Pi 4, 4GB. Official Pi camera module v2 attached.

When I point my browser at the device public URL (https://xxx.balena-devices.com) the page loads and all seems to be going OK, until my browser looks like this with no video:

[image]

The text 'Double tap the video to enter fullscreen mode!' pops up if I move the mouse around.

In the device logs I see the following:

```
Supervisor starting
<balena-cam> Video device is ready
<balena-cam> 
<balena-cam> #############################################################
<balena-cam> Authorization is disabled.
<balena-cam> Anyone can access your balenaCam, using the device's URL!
<balena-cam> Set the username and password environment variables 
<balena-cam> to enable authorization.
<balena-cam> For more info visit: 
<balena-cam> https://github.com/balena-io-playground/balena-cam
<balena-cam> #############################################################
<balena-cam> 
<balena-cam> ======== Running on http://0.0.0.0:80 ========
<balena-cam> (Press CTRL+C to quit)
<balena-cam> Task exception was never retrieved
<balena-cam> future: <Task finished coro=<RTCPeerConnection.__connect() done, defined at /usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py:554> exception=ValueError("'da65f2fd-7b20-4bdd-b0fb-6a472657d77d.local' does not appear to be an IPv4 or IPv6 address",)>
<balena-cam> Traceback (most recent call last):
<balena-cam>   File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
<balena-cam>     result = coro.send(None)
<balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py", line 559, in __connect
<balena-cam>     await iceTransport.start(self.__remoteIce[transceiver])
<balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcicetransport.py", line 284, in start
<balena-cam>     await self._connection.connect()
<balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/ice.py", line 356, in connect
<balena-cam>     if (protocol.local_candidate.can_pair_with(remote_candidate) and
<balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/candidate.py", line 114, in can_pair_with
<balena-cam>     b = ipaddress.ip_address(other.host)
<balena-cam>   File "/usr/lib/python3.5/ipaddress.py", line 54, in ip_address
<balena-cam>     address)
<balena-cam> ValueError: 'da65f2fd-7b20-4bdd-b0fb-6a472657d77d.local' does not appear to be an IPv4 or IPv6 address
```

I suspect that error message is due to https://github.com/aiortc/aiortc/issues/187. As far as I can tell it's unlikely to be causing the problem I see.

I've spent time trying to rebuild this project with newer versions of aioice and other packages, but failed to find a combination of versions which will build happily.

My browser is 'Chrome Version 83.0.4103.106 (Official Build) (64-bit)'. Here's the browser console output from visiting the page:

```
client.js:112 Offer SDP
client.js:113 v=0
o=- 4261409655020767845 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=msid-semantic: WMS
m=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 122 127 121 125 107 108 109 124 120 123
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=candidate:3199543849 1 udp 2113937151 a4fa2528-a9f8-4f30-b479-6991d9101a71.local 48065 typ host generation 0 network-cost 999
a=candidate:559844702 1 udp 2113939711 bce860ed-e1d0-4ad8-810f-0ef967ee58c2.local 36761 typ host generation 0 network-cost 999
a=ice-ufrag:BNW/
a=ice-pwd:JEp9VV3l5qs66xwHVR8Df6Ie
a=ice-options:trickle
a=fingerprint:sha-256 07:F5:B1:03:C5:75:22:01:E2:F1:39:FC:B7:E1:F8:ED:87:6A:E9:7A:DC:0B:B8:C2:62:17:BC:D9:41:1E:49:CE
a=setup:actpass
a=mid:0
a=extmap:14 urn:ietf:params:rtp-hdrext:toffset
a=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:13 urn:3gpp:video-orientation
a=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
a=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
a=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
a=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07
a=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space
a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid
a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
a=extmap:6 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
a=recvonly
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:96 VP8/90000
a=rtcp-fb:96 goog-remb
a=rtcp-fb:96 transport-cc
a=rtcp-fb:96 ccm fir
a=rtcp-fb:96 nack
a=rtcp-fb:96 nack pli
a=rtpmap:97 rtx/90000
a=fmtp:97 apt=96
a=rtpmap:98 VP9/90000
a=rtcp-fb:98 goog-remb
a=rtcp-fb:98 transport-cc
a=rtcp-fb:98 ccm fir
a=rtcp-fb:98 nack
a=rtcp-fb:98 nack pli
a=fmtp:98 profile-id=0
a=rtpmap:99 rtx/90000
a=fmtp:99 apt=98
a=rtpmap:100 VP9/90000
a=rtcp-fb:100 goog-remb
a=rtcp-fb:100 transport-cc
a=rtcp-fb:100 ccm fir
a=rtcp-fb:100 nack
a=rtcp-fb:100 nack pli
a=fmtp:100 profile-id=2
a=rtpmap:101 rtx/90000
a=fmtp:101 apt=100
a=rtpmap:102 H264/90000
a=rtcp-fb:102 goog-remb
a=rtcp-fb:102 transport-cc
a=rtcp-fb:102 ccm fir
a=rtcp-fb:102 nack
a=rtcp-fb:102 nack pli
a=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f
a=rtpmap:122 rtx/90000
a=fmtp:122 apt=102
a=rtpmap:127 H264/90000
a=rtcp-fb:127 goog-remb
a=rtcp-fb:127 transport-cc
a=rtcp-fb:127 ccm fir
a=rtcp-fb:127 nack
a=rtcp-fb:127 nack pli
a=fmtp:127 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42001f
a=rtpmap:121 rtx/90000
a=fmtp:121 apt=127
a=rtpmap:125 H264/90000
a=rtcp-fb:125 goog-remb
a=rtcp-fb:125 transport-cc
a=rtcp-fb:125 ccm fir
a=rtcp-fb:125 nack
a=rtcp-fb:125 nack pli
a=fmtp:125 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f
a=rtpmap:107 rtx/90000
a=fmtp:107 apt=125
a=rtpmap:108 H264/90000
a=rtcp-fb:108 goog-remb
a=rtcp-fb:108 transport-cc
a=rtcp-fb:108 ccm fir
a=rtcp-fb:108 nack
a=rtcp-fb:108 nack pli
a=fmtp:108 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42e01f
a=rtpmap:109 rtx/90000
a=fmtp:109 apt=108
a=rtpmap:124 red/90000
a=rtpmap:120 rtx/90000
a=fmtp:120 apt=124
a=rtpmap:123 ulpfec/90000

client.js:127 Answer SDP
client.js:128 v=0
o=- 3802163393 3802163393 IN IP4 0.0.0.0
s=-
t=0 0
a=group:BUNDLE 0
m=video 51929 UDP/TLS/RTP/SAVPF 96 102 125
c=IN IP4 192.168.1.208
a=sendonly
a=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid
a=mid:0
a=rtcp:9 IN IP4 0.0.0.0
a=rtcp-mux
a=ssrc:3960971726 cname:{6d8b996c-925a-4b17-8130-a26bc0da478d}
a=rtpmap:96 VP8/90000
a=rtcp-fb:96 nack
a=rtcp-fb:96 nack pli
a=rtcp-fb:96 goog-remb
a=rtpmap:102 H264/90000
a=rtcp-fb:102 nack
a=rtcp-fb:102 nack pli
a=rtcp-fb:102 goog-remb
a=fmtp:102 profile-level-id=42001f;packetization-mode=1;level-asymmetry-allowed=1
a=rtpmap:125 H264/90000
a=rtcp-fb:125 nack
a=fmtp:125 profile-level-id=42e01f;packetization-mode=1;level-asymmetry-allowed=1
a=candidate:1ca67abb0de160439e2b97a25cbca3d3 1 udp 2130706431 192.168.1.208 51929 typ host
a=candidate:cf50604904dba48bcca6c33a0952139c 1 udp 2130706431 2a00:23c7:682:e601:5df2:5d28:5982:6e67 60504 typ host
a=candidate:6023c2604ec050669e1ed7b3429afd22 1 udp 2130706431 10.114.102.1 65148 typ host
a=candidate:83f0d7982cc74ffd6cf0805b1c70b14f 1 udp 2130706431 10.114.101.1 64902 typ host
a=candidate:98878514fb0b951fbaeec78e059c2f85 1 udp 2130706431 10.114.104.1 56527 typ host
a=candidate:580b5f32035da7f14a30ea7a8d826c67 1 udp 2130706431 172.17.0.1 52483 typ host
a=candidate:3f746d0a55efc1da799b1fcd969c908c 1 udp 2130706431 10.240.52.203 59097 typ host
a=candidate:ea2179218112e69c0463eae46ff5f653 1 udp 1694498815 109.158.209.172 64902 typ srflx raddr 10.114.101.1 rport 64902
a=candidate:401429fb19a72255c8ce07f92fe227f9 1 udp 1694498815 109.158.209.172 56527 typ srflx raddr 10.114.104.1 rport 56527
a=candidate:0ad8db753210ce9554cf251a9e51e792 1 udp 1694498815 109.158.209.172 52483 typ srflx raddr 172.17.0.1 rport 52483
a=candidate:c2ed111e16f11519afec0b3650ccc346 1 udp 1694498815 109.158.209.172 51929 typ srflx raddr 192.168.1.208 rport 51929
a=end-of-candidates
a=ice-ufrag:yZJE
a=ice-pwd:vU87PjBw1nH2ZCgpo6AHWf
a=fingerprint:sha-256 B9:42:0E:E7:D9:7B:98:3F:0A:90:C1:E7:24:E1:F9:36:BD:3C:43:0E:1B:A0:13:04:00:E4:6D:94:58:B0:AC:FA
a=setup:active

client.js:64 checking
mainIceListener @ client.js:64
client.js:64 connected
mainIceListener @ client.js:64
client.js:25 Attaching stream...
```

I have tried one other thing, which is to force mjpeg mode in client.js:

```
- if (window.navigator.userAgent.indexOf("Edge") > -1  || safariOnIos) {
+ if (true) {
```

This works: I see mjpeg video in the browser. So I suspect the problem is to do specifically with WebRTC.

mbalamat, 2020-07-06:
@HughWarrington see https://github.com/balenalabs/balena-cam#additional-information on the 4th bullet about chrome. It will resolve this for you. Sorry for the late reply, FYI we respond very quickly on our [forums](https://forums.balena.io/). I'll close this for now. Feel free to re-open it if it doesn't work for you.

closed 2020-07-06 by mbalamat

HughWarrington, 2020-07-09:
Thank you, that does indeed fix the problem! (after I fixed the build, see #74)

A small improvement would be to detect the user is running Chrome and show a link to the relevant part of the docs.

HughWarrington, 2020-07-09:
Documentation improvement submitted in #75

chilipeppr, 2020-10-18:
FYI, you have to toggle this setting to get to the suggested setting in #75 

[image]