balena-labs-projects/balena-cam #74: Cannot build: No matching distribution found for av<7.0.0,>=0.5.3 (from aiortc==0.9.11)

HughWarrington, 2020-07-09:
I hit the big 'Deploy with balena' button on https://github.com/balenalabs/balena-cam/blob/master/README.md, and this is what happened:

```
Step 1/9 : FROM balenalib/raspberrypi3-debian:stretch-20190612
 ---> 9a857048cc34
Step 2/9 : RUN apt-get update &&   apt-get install -yq     python3     python3-dev     python3-pip     python3-setuptools     gstreamer-1.0     v4l-utils     libopus-dev     libvpx-dev     libsrtp2-dev     libopencv-dev     libatlas3-base     libatlas-base-dev     libjasper-dev     libilmbase12     libopenexr22     libavformat-dev     libswscale-dev     libqtgui4     libqt4-test     libavdevice-dev     libavfilter-dev     libavcodec-dev   && apt-get clean && rm -rf /var/lib/apt/lists/*
Using cache
 ---> 050a958f4078
Step 3/9 : RUN pip3 install --upgrade pip
Using cache
 ---> 03e352a70ab7
Step 4/9 : RUN pip3 install async-timeout
Using cache
 ---> 02ece5446493
Step 5/9 : RUN pip3 install aiohttp aiohttp_basicauth==0.1.3 aioice==0.6.10 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple
 ---> Running in 5e69f8b59472
Looking in indexes: https://www.piwheels.org/simple, https://www.piwheels.org/simple
Collecting aiohttp
  Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-3.6.2-py3-none-any.whl (441 kB)
Collecting aiohttp_basicauth==0.1.3
  Downloading https://www.piwheels.org/simple/aiohttp-basicauth/aiohttp_basicauth-0.1.3-py3-none-any.whl (7.0 kB)
Collecting aioice==0.6.10
  Downloading https://www.piwheels.org/simple/aioice/aioice-0.6.10-py3-none-any.whl (19 kB)
Collecting aiortc==0.9.11
  Downloading https://www.piwheels.org/simple/aiortc/aiortc-0.9.11-cp35-cp35m-linux_armv7l.whl (112 kB)
Collecting numpy==1.15.4
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.15.4-cp35-cp35m-linux_armv7l.whl (6.4 MB)
Collecting opencv-python==3.4.4.19
  Downloading https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.4.19-cp35-cp35m-linux_armv7l.whl (8.8 MB)
Collecting multidict<5.0,>=4.5
  Downloading https://www.piwheels.org/simple/multidict/multidict-4.7.6-cp35-cp35m-linux_armv7l.whl (139 kB)
Requirement already satisfied: async-timeout<4.0,>=3.0 in /usr/local/lib/python3.5/dist-packages (from aiohttp) (3.0.1)
Collecting attrs>=17.3.0
  Downloading https://www.piwheels.org/simple/attrs/attrs-19.3.0-py2.py3-none-any.whl (39 kB)
Collecting idna-ssl>=1.0; python_version < "3.7"
  Downloading https://www.piwheels.org/simple/idna-ssl/idna_ssl-1.1.0-py3-none-any.whl (2.3 kB)
Collecting yarl<2.0,>=1.0
  Downloading https://www.piwheels.org/simple/yarl/yarl-1.4.2-cp35-cp35m-linux_armv7l.whl (234 kB)
Collecting typing-extensions>=3.6.5; python_version < "3.7"
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-3.7.4.2-py3-none-any.whl (22 kB)
Collecting chardet<4.0,>=2.0
  Downloading https://www.piwheels.org/simple/chardet/chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting netifaces
  Downloading https://www.piwheels.org/simple/netifaces/netifaces-0.10.9-cp35-cp35m-linux_armv7l.whl (40 kB)
Collecting crcmod
  Downloading https://www.piwheels.org/simple/crcmod/crcmod-1.7-cp35-cp35m-linux_armv7l.whl (37 kB)
Collecting pylibsrtp>=0.5.6
  Downloading https://www.piwheels.org/simple/pylibsrtp/pylibsrtp-0.6.0-cp35-cp35m-linux_armv7l.whl (25 kB)
Collecting pyee
  Downloading https://www.piwheels.org/simple/pyee/pyee-7.0.2-py2.py3-none-any.whl (12 kB)
ERROR: Could not find a version that satisfies the requirement av<7.0.0,>=0.5.3 (from aiortc==0.9.11) (from versions: none)

ERROR: No matching distribution found for av<7.0.0,>=0.5.3 (from aiortc==0.9.11)

Removing intermediate container 5e69f8b59472
The command '/bin/sh -c pip3 install aiohttp aiohttp_basicauth==0.1.3 aioice==0.6.10 aiortc==0.9.11 numpy==1.15.4 opencv-python==3.4.4.19 --index-url https://www.piwheels.org/simple' returned a non-zero code: 1
```

I see that https://www.piwheels.org/simple/av is completely empty. Is that a 'bug' by piwheels? I'm not sure how to tell.

My thinking was 'well there's no pre-built wheel, never mind, I'll just build it myself'. So I changed line 34 of the Dockerfile from `--index-url` to `--extra-index-url`. Rather than failing when it can't find a ready-build wheel for av, it tries to built the package.

```
[balena-cam]  Using legacy setup.py install for pylibsrtp, since package 'wheel' is not installed.
[balena-cam]  Using legacy setup.py install for av, since package 'wheel' is not installed.
```

It also tried to build pylibsrtp, which failed, but that is fixed by adding libffi-dev to the apt-get install list earlier in the Dockerfile.

I'm not sure if the fix I've made is the right one, since I'm unsure of the rationale behind the existing series of pip commands. In particular, I'm not sure why `pip3 install av` is a separate line in the Dockerfile, since aiortc already depends on av. If my fix is considered a win though, let me know and I'll open a PR.

mbalamat, 2020-07-09:
@HughWarrington can you try the branch from this [PR](https://github.com/balenalabs/balena-cam/pull/76) it locks av to v6.1.0. The rationale behind the different lines is that it uses PyPI, piwheels and then PyPI again, in the point of failure it used piwheels when it should have used PyPI. Your fix is considered a win and we are open for PRs in general, but if we don't need to build something which takes time, it's even better. If you can make balenaCam work using newer base images and dependencies or add a feature like saving video/images on the device feel free to send PRs our way, we would greatly appreciate your contribution.

HughWarrington, 2020-07-14:
> can you try the branch from this PR it locks av to v6.1.0

I just built and deployed it successfully on my hardware (Pi4, Pi camera module v2), and WebRTC is working great in my browser (Chrome, after following steps in the README).

I would consider that PR a fix to this ticket. Many thanks!

mbalamat, 2020-07-14:
@HughWarrington great and thank you for your docs PR!

closed 2020-07-14 by mbalamat