balena-labs-projects/balena-cam #78: Show date and time in stream

leonpierre, 2020-09-18:
Is this possible?

mbalamat, 2020-11-01:
Hello @leonpierre yes it is. In general, balenaCam uses opencv to get the frame from the camera. So you can put whatever you want on a frame, you can even apply CV algorithms before returning it. WRT to your question, a user asked the same in our forums I can't find the forums post but I found the fork that this is implemented in, here is the diff that does this https://github.com/shams3049/balena-cam/commit/b0ca3845a4bb62f213f3e21c3cb43d3670315793.

mbalamat, 2020-11-01:
Also, feel free to send us a PR with this addition, here. I am closing this, for now, feel free to re-open it if you have any issues.

closed 2020-11-01 by mbalamat