balena-labs-projects/balena-cam #86: "Blank" screen in web browser

aweussom, 2020-11-09:
Hi. My balenacam is not working. When I try to access the WebRTC stream, everything but the stream itself loads.
I can double-click to go fullscreen etc - but no video.
The camera and RPI 3 A+ is confirmed working.

From the logs - the only weird thing is the .local address instead of the IP my computer is on?

----
`09.11.20 12:20:45 (+0100) <balena-cam> https://github.com/balena-io-playground/balena-cam
09.11.20 12:20:45 (+0100) <balena-cam> #############################################################
09.11.20 12:20:45 (+0100) <balena-cam> 
09.11.20 12:20:45 (+0100) <balena-cam> ======== Running on http://0.0.0.0:80 ========
09.11.20 12:20:45 (+0100) <balena-cam> (Press CTRL+C to quit)
09.11.20 12:21:00 (+0100) <balena-cam> /usr/local/lib/python3.5/dist-packages/pyee/_base.py:81: RuntimeWarning: coroutine 'on_iceconnectionstatechange' was never awaited
09.11.20 12:21:00 (+0100) <balena-cam>   f(*args, **kwargs)
09.11.20 12:21:00 (+0100) <balena-cam> Task exception was never retrieved
09.11.20 12:21:00 (+0100) <balena-cam> future: <Task finished coro=<RTCPeerConnection.__connect() done, defined at /usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py:554> exception=ValueError("'2e20163f-0920-4908-b6b7-f6e3b8ce882a.local' does not appear to be an IPv4 or IPv6 address",)>
09.11.20 12:21:00 (+0100) <balena-cam> Traceback (most recent call last):
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
09.11.20 12:21:00 (+0100) <balena-cam>     result = coro.send(None)
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcpeerconnection.py", line 559, in __connect
09.11.20 12:21:00 (+0100) <balena-cam>     await iceTransport.start(self.__remoteIce[transceiver])
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aiortc/rtcicetransport.py", line 284, in start
09.11.20 12:21:00 (+0100) <balena-cam>     await self._connection.connect()
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/ice.py", line 356, in connect
09.11.20 12:21:00 (+0100) <balena-cam>     if (protocol.local_candidate.can_pair_with(remote_candidate) and
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/local/lib/python3.5/dist-packages/aioice/candidate.py", line 114, in can_pair_with
09.11.20 12:21:00 (+0100) <balena-cam>     b = ipaddress.ip_address(other.host)
09.11.20 12:21:00 (+0100) <balena-cam>   File "/usr/lib/python3.5/ipaddress.py", line 54, in ip_address
09.11.20 12:21:00 (+0100) <balena-cam>     address)
09.11.20 12:21:00 (+0100) <balena-cam> ValueError: '2e20163f-0920-4908-b6b7-f6e3b8ce882a.local' does not appear to be an IPv4 or IPv6 address`

aweussom, 2020-11-09:
[3dprintercam_alfheim-prusa-mini-09.11.20_15 16 19.txt](https://github.com/balenalabs/balena-cam/files/5511029/3dprintercam_alfheim-prusa-mini-09.11.20_15.16.19.txt)

Attached log

aweussom, 2020-11-09:
Looks like some sort of Bonjour-related problem?
Also tested on my Cellphone wifi sharing, to remove Bonjour from the equation. Same problem

mbalamat, 2020-11-09:
@aweussom balenaCam doesn't work with mdns adresses. In order to disable them see here https://github.com/balenalabs/balena-cam#additional-information.

aweussom, 2020-11-11:
That option is not available in my Chrome.
I blocked mDNS with the windows 10 firewall, and that worked fine:
[In a dministrator powershell]

Set-NetFirewallRule -DisplayName "Google Chrome (mDNS-In)" -Enabled True -Action Block