balena-labs-projects/balena-cam #99: Doesn't run successfully when pushed to balena; syntax error

mozz100, 2022-02-02:
From the current `master` branch, I cloned and then pushed to balena.  Image built ok and has deployed to my device (raspi 4)

However the `balena-cam` container is failing to start, falling over with a syntax error.  Seems to be complaining about type hinting in pyee.  Has some dependency not been pinned and an external release rendered this code un-runnable?

```
Traceback (most recent call last):
    File "/usr/src/app/server.py", line 5, in <module>
      from aiortc import RTCPeerConnection, RTCSessionDescription, VideoStreamTrack, RTCIceServer, RTCConfiguration
    File "/usr/local/lib/python3.5/dist-packages/aiortc/__init__.py", line 2, in <module>
      from .mediastreams import (AudioStreamTrack,  # noqa
    File "/usr/local/lib/python3.5/dist-packages/aiortc/mediastreams.py", line 7, in <module>
      from pyee import EventEmitter
    File "/usr/local/lib/python3.5/dist-packages/pyee/__init__.py", line 34, in <module>
      from pyee.base import EventEmitter as EventEmitter
    File "/usr/local/lib/python3.5/dist-packages/pyee/base.py", line 43
      self._events: Dict[
                  ^
  SyntaxError: invalid syntax
```

zoobot, 2022-03-08:
same issue for me.

mozz100, 2022-03-09:
@zoobot I found that pinning pyee (as in https://github.com/balenalabs/balena-cam/pull/101) allowed me to work around this and get started.  Not sure if @balenalabs are still maintaining this code?

mozz100, 2022-03-29:
This should be fixed in v0.2.9; marking closed

closed 2022-03-29 by mozz100